BojanJurca/Esp32_oscilloscope #1: file_system.h:27:12: fatal error: SPIFFS.h: No such file or directory

rlfnb, 2020-08-29:
cloned your repo and tried to compile, running into this 
`
In file included from /home/poc/workspace/Esp32_oscilloscope/Esp32_oscilloscope.ino:26:0:
file_system.h:27:12: fatal error: SPIFFS.h: No such file or directory
   #include <SPIFFS.h>  
            ^~~~~~~~~~
compilation terminated.
exit status 1
SPIFFS.h: No such file or directory
`
any idea?

BojanJurca, 2020-08-30:
Hi, it must be a problem with different versions of different files I have uploaded through time. I'll look into it ASAP. 

If you can't wait please download Esp32_web_ftp_telnet_server_template (https://github.com/BojanJurca/Esp32_web_ftp_telnet_server_template), Oscilloscope is also included there with all the latest versions of all files.

BojanJurca, 2020-08-30:
I have just uploaded the latest version of all files which are now available for download. Sketch compiles OK. I still have to do some testing before publishing a new release.

BojanJurca, 2020-08-30:
... by the way ... do you compile the sketch for one of ESP32 boards? Oscilloscope only works on ESP32.

rlfnb, 2020-09-01:
thank you a lot for taking care of the ticket. the SPIFFS issue is gone, but I got another one. Will open a new one for it.
Im using the ESP32 DevKitC_v4.

closed 2020-09-01 by rlfnb