BojanJurca/Esp32_oscilloscope #2: stackoverflow in TcpConnection

rlfnb, 2020-09-01:
made the build, uploaded it with my wifi credentials changed before and run into the stackoverflow, when i try to connect to port 80, maybe you can see it already? If you need more information let me know!

'
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[        32] [network] [STA] connecting STAtion to router using DHCP ...
[       105] [network] WiFi interface ready.
[       105] [network] [STA] WiFi client started.
[       106] [file system] mounting SPIFFS ...
[       202] [file system] SPIFFS mounted.
[       204] [ftpServer] started on 0.0.0.0:21.
[       204] FTP server has started.
[       215] [httpServer] started on 0.0.0.0:80.
[       215] WEB server has started.
[       251] [network] [STA] connected to WiFi **************.
[       260] [network] [STA] obtained IP address nnn.nnn.nnn.nnn.
***ERROR*** A stack overflow in task TcpConnection has been detected.
abort() was called at PC 0x4008ca48 on core 1

Backtrace: 0x4008c800:0x3ffce530 0x4008ca31:0x3ffce550 0x4008ca48:0x3ffce570 0x40089b14:0x3ffce590 0x4008b778:0x3ffce5b0 0x4008b72e:0xfefefefe

Rebooting...
'

BojanJurca, 2020-09-01:
I added some additional stack memory to httpServer in Esp32_oscilloscope.ino. It shoud work now.

rlfnb, 2020-09-01:
everything's working now on DevKitC_V4! Thanks a lot!

closed 2020-09-01 by rlfnb