BojanJurca/Esp32_oscilloscope #4: WebSocket error: [object Event]

pierrebak, 2020-12-20:
Hello,

Thanks for the nice project.
I've been able to set it up, but when trying to start monitoring on a GPIO, i have an error message: WebSocket error: [object Event]

Any idea on how to fix it ?

thanks

BojanJurca, 2021-01-04:
Hello, I'm sorry for late reply, I somehow missed this issue. Please tell me more about what you are doing. Are you running oscilloscope on the same board as PWM you want to monitor or do you use two different boards?

BojanJurca, 2021-01-04:
Oscilloscope was meant to monitor signals on the same ESP32 as your projects runs in the first place. Of course it can also be used to monitor external devices. Before using Oscilloscope you need to initialize pin mode of the pin that you are going to use by Oscilloscope. You can do this in Oscilloscope Arduino code (in setup () for example).

BojanJurca, 2021-01-04:
Example:

void setup () {  
  Serial.begin (115200);

  /*
    ...
  */

  ledcSetup (0, 83, 10);  // channel 0, 83 Hz , 10 bit resolution (1024 values)
  ledcAttachPin (2, 0);   // attach pin 2 (built-in led) to channel 0

  ledcWrite (0, 307);     // PWM of built-in LED (channel 0 -> pin 2) with 1/3 of period (307 out of 1023)    
}

Then you can see:

[image]

BojanJurca, 2021-01-05:
I'm coding this issue. Please reopen it if this didn't solve your problem.

closed 2021-01-05 by BojanJurca