BojanJurca/Esp32_oscilloscope #6: Request for barebone websocket server

happytm, 2021-05-13:
This is not an issue.

Thank you for comprehensive multi web server piece of software. I am enjoying testing it.

If time allows can you just create a repository  with just websocket server for ESP32?  

No FTP server, no HTTP server , no time related code. Just plain websocket server which can communicate with javascript client in web browser and may be display a graph from couple of variables like you show for freeheap and uptime on  index page of  your Esp32_web_ftp_telnet_server_template.

It should be  barebone template where it is simple and people like me  can understand  and use it in their small project like home automation without complexity of all in one software.

Thank you.

BojanJurca, 2021-06-07:
Hi. Sorry for late reply, I was taking vacations. I'll prepare something for you soon.

BojanJurca, 2021-06-07:
/*

    Demo1 - this is a minimum you need to do in order to make (only) web server (without websockets) to work

    (other demos will follow)
 
*/

#include <WiFi.h>

#define HOSTNAME    "MyESP32Server" // define the name of your ESP32 here
#define MACHINETYPE "ESP32 NodeMCU" // describe your hardware here

// connect ESP32 in STAtion mode to your WiFi router ...
#define DEFAULT_STA_SSID          "YOUR_STA_SSID"
#define DEFAULT_STA_PASSWORD      "YOUR_STA_PASSWORD"
// ... but don't let ESP32 to set up its own A(ccess) P(oit)
#define DEFAULT_AP_SSID           ""
#define DEFAULT_AP_PASSWORD       ""

#include "./servers/network.h"
#include "./servers/webServer.hpp"

// Callback function that will handle all HTTP requests (since we do not have any file system mounted
// to be able to pass .html and other files to client, this callback function is the only possibility 
// we have to respond to HTTP requests from clients (browsers)
 
String httpRequestHandler (String& httpRequest, httpServer::wwwSessionParameters *wsp) { // - must be reentrant!
  if (httpRequest.substring (0, 6) == "GET / ") { 
    return String ("<HTML>Example 01 - dynamic HTML page<br><br><hr />") + (digitalRead (2) ? "Led is on." : "Led is off.") + String ("<hr /></HTML>");
  }

  return ""; // httpRequestHandler did not handle the request - tell httpServer to handle it internally (404, ...) by returning "" reply
}

void setup () {
  Serial.begin (115200);
 
  startNetworkAndInitializeItAtFirstCall ();                          // since we do not have any file system mounted this would only start WiFi
                                                                      // but won't initialize network configuration files

  // start web server 
  httpServer *httpSrv = new httpServer (httpRequestHandler,           // our HTTP callback function 
                                        NULL,                         // no WS callback function
                                        8 * 1024,                     // 8 KB stack size is usually enough, if httpRequestHandler or wsRequestHandler use more stack increase this value until server is stable
                                        (char *) "0.0.0.0",           // start HTTP server on all available ip addresses
                                        80,                           // HTTP port
                                        NULL);                        // no firewall callback function
  if (!httpSrv || (httpSrv && !httpSrv->started ())) Serial.println ("[httpServer] did not start.");
}

void loop () {

}

BojanJurca, 2021-06-07:
The function that is not displayed correctly looks like this:

[image]

BojanJurca, 2021-06-07:
/*

    Demo2 - this is a minimum you need to do in order to make (only) web server with websockets to work

    (other demos will follow)
 
*/

#include <WiFi.h>

#define HOSTNAME    "MyESP32Server" // define the name of your ESP32 here
#define MACHINETYPE "ESP32 NodeMCU" // describe your hardware here

// connect ESP32 in STAtion mode to your WiFi router ...
#define DEFAULT_STA_SSID          "YOUR_STA_SSID"
#define DEFAULT_STA_PASSWORD      "YOUR_STA_PASSWORD"
// ... but don't let ESP32 to set up its own A(ccess) P(oit)
#define DEFAULT_AP_SSID           ""
#define DEFAULT_AP_PASSWORD       ""

#include "./servers/network.h"
#include "./servers/webServer.hpp"

// Callback function that will handle all HTTP requests (since we do not have any file system mounted
// to be able to pass .html and other files to client, this callback function is the only possibility 
// we have to respond to HTTP requests from clients (browsers)
 
String httpRequestHandler (String& httpRequest, httpServer::wwwSessionParameters *wsp) { // - must be reentrant!
  if (httpRequest.substring (0, 6) == "GET / ") { 
    return "<html>\n"
           "  <body>\n"
           "    RSSI <div id='rssi'>...</div> dBm\n"
           "\n" 
           "    <script type='text/javascript'>\n"
           "      if ('WebSocket' in window) {\n"
           "        var ws = new WebSocket ('ws://' + self.location.host + '/streamRSSI'); // open webSocket connection\n"
           "        ws.onopen = function () {;};\n"
           "        ws.onmessage = function (evt) {\n" 
           "          if (evt.data instanceof Blob) { // RSSI readings as 1 byte binary data\n"
           "            // receive binary data as blob and then convert it into array\n"
           "            var myByteArray = null;\n"
           "            var myArrayBuffer = null;\n"
           "            var myFileReader = new FileReader ();\n"
           "            myFileReader.onload = function (event) {\n"
           "              myArrayBuffer = event.target.result;\n"
           "              myByteArray = new Int8Array (myArrayBuffer); // <= RSSI information is now in the array of size 1\n"
           "\n"
           "              // this is where you pass a byte received through websocket into a GUI control\n"
           "              document.getElementById('rssi').innerText = myByteArray [0].toString ();\n"
           "\n"
           "            };\n"
           "            myFileReader.readAsArrayBuffer (evt.data);\n"
           "          }\n"
           "        };\n"
           "      } else {\n"
           "        alert ('WebSockets are not supported by your browser.');\n"
           "      }\n"
           "    </script>\n"
           "  </body>\n"
           "</html>\n";
  }

  return ""; // httpRequestHandler did not handle the request - tell httpServer to handle it internally (404, ...) by returning "" reply
}

// Callback function that will handle all WS (websocket) requests - they are always called by web browser from HTML (responses or files - see above)

void wsRequestHandler (String& wsRequest, WebSocket *webSocket) { // - must be reentrant!
  if (wsRequest.substring (0, 16) == "GET /streamRSSI ") {
    char c;
    do {
      delay (1000);
      int i = WiFi.RSSI ();
      c = (char) i;
    } while (webSocket->sendBinary ((byte *) &c,  sizeof (c))); // send RSSI information as long as web browser is willing to receive it
  }
}

void setup () {
  Serial.begin (115200);
 
  startNetworkAndInitializeItAtFirstCall ();                          // since we do not have any file system mounted this would only start WiFi
                                                                      // but won't initialize network configuration files

  // start web server 
  httpServer *httpSrv = new httpServer (httpRequestHandler,           // our HTTP callback function 
                                        wsRequestHandler,             // our WS callback function
                                        8 * 1024,                     // 8 KB stack size is usually enough, if httpRequestHandler or wsRequestHandler use more stack increase this value until server is stable
                                        (char *) "0.0.0.0",           // start HTTP server on all available ip addresses
                                        80,                           // HTTP port
                                        NULL);                        // no firewall callback function
  if (!httpSrv || (httpSrv && !httpSrv->started ())) Serial.println ("[httpServer] did not start.");
}

void loop () {

}

BojanJurca, 2021-06-07:
/*

    Demo3 - mount FAT file system - it is easier to put HTML responses into files than into C variables or literals
            select Tools | Partition scheme | one of FAT partition sachems
          - we will also need FTP server in order to upload HTML file onto ESP32 FAT file system 
            
    (other demos will follow)

*/

#include <WiFi.h>

#define HOSTNAME    "MyESP32Server" // define the name of your ESP32 here
#define MACHINETYPE "ESP32 NodeMCU" // describe your hardware here

// connect ESP32 in STAtion mode to your WiFi router ...
#define DEFAULT_STA_SSID          "YOUR_STA_SSID"
#define DEFAULT_STA_PASSWORD      "YOUR_STA_PASSWORD"
// ... but don't let ESP32 to set up its own A(ccess) P(oit)
#define DEFAULT_AP_SSID           ""
#define DEFAULT_AP_PASSWORD       ""

#include "./servers/file_system.h"
#include "./servers/network.h"

#define USER_MANAGEMENT NO_USER_MANAGEMENT  // although NO_USER_MANAGEMENT is selected we still need user_management.h - this is where
#include "./servers/user_management.h"      // web server will get its root directory from, whilest FTP server will accept whatever you type in

#include "./servers/webServer.hpp"
#include "./servers/ftpServer.hpp"

// httpRequestHandler HTTP callback function is no longer needed - web server can read its HTTP response from a HTML file
 
// Callback function that will handle all WS (websocket) requests - they are always called by web browser from HTML (responses or files - see above)

void wsRequestHandler (String& wsRequest, WebSocket *webSocket) { // - must be reentrant!
  if (wsRequest.substring (0, 16) == "GET /streamRSSI ") {
    char c;
    do {
      delay (1000);
      int i = WiFi.RSSI ();
      c = (char) i;
    } while (webSocket->sendBinary ((byte *) &c,  sizeof (c))); // send RSSI information as long as web browser is willing to receive it
  }
}

void setup () {
  Serial.begin (115200);

  // FFat.format ();                                                    // delete all files if you need to start from scratch
  mountFileSystem (true);                                             // mount (and format if neccessary) file system
 
  startNetworkAndInitializeItAtFirstCall ();                          // initialize network configuration files and start network

  // start web server 
  httpServer *httpSrv = new httpServer (NULL,                         // no HTTP callback function 
                                        wsRequestHandler,             // our WS callback function
                                        8 * 1024,                     // 8 KB stack size is usually enough, if httpRequestHandler or wsRequestHandler use more stack increase this value until server is stable
                                        (char *) "0.0.0.0",           // start HTTP server on all available ip addresses
                                        80,                           // HTTP port
                                        NULL);                        // no firewall callback function
  if (!httpSrv || (httpSrv && !httpSrv->started ())) Serial.println ("[httpServer] did not start.");

  // start FTP server
  ftpServer *ftpSrv = new ftpServer ("0.0.0.0",                       // start FTP server on all available ip addresses
                                     21,                              // controll connection FTP port
                                     NULL);                           // no firewall callback function
  if (!ftpSrv || (ftpSrv && !ftpSrv->started ())) Serial.println ("[ftpServer] did not start.");  
}

void loop () {

}

After ESP32 starts you will have to FTP to it and upload index.html into /var/www/html directory (just type anything for username and password). index.html would look like this:

<html>
  <body>
    RSSI <div id='rssi'>...</div> dBm

    <script type='text/javascript'>
      if ('WebSocket' in window) {
        var ws = new WebSocket ('ws://' + self.location.host + '/rssiReader'); // open webSocket connection
        ws.onopen = function () {;};
        ws.onmessage = function (evt) { 
          if (evt.data instanceof Blob) { // RSSI readings as 1 byte binary data
            // receive binary data as blob and then convert it into array¸
            var myByteArray = null;
            var myArrayBuffer = null;
            var myFileReader = new FileReader ();
            myFileReader.onload = function (event) {
              myArrayBuffer = event.target.result;
              myByteArray = new Int8Array (myArrayBuffer); // <= RSSI information is now in the array of size 1

              // this is where you pass a byte received through websocket into a GUI control
              document.getElementById('rssi').innerText = myByteArray [0].toString ();

            };
            myFileReader.readAsArrayBuffer (evt.data);
          }
        };
      } else {
        alert ('WebSockets are not supported by your browser.');
      }
    </script>
  </body>
</html>

BojanJurca, 2021-06-07:
Some tags have disappeared - I'll just upload ZIP file.
[Demo3.zip](https://github.com/BojanJurca/Esp32_oscilloscope/files/6609095/Demo3.zip)

BojanJurca, 2021-06-07:
Everything else is packing data into stream of bytes on the server side (ESP32) and parsing the same stream on the client side (javascript in web browser). Now suppose we want to stream samples from A/D converters on pins 36 and 39 each 1/10th of a second. Each sample fits into 16 bit unsigned integer. Therefore each 1/10th of a second ESP32 will push 2, 16 bit unsigned integers (4 bytes) into websocket which will be read by javascript in web browser and parsed into 2 16 bit unsigned integers. 

/*

    Demo4

*/

#include <WiFi.h>

#define HOSTNAME    "MyESP32Server" // define the name of your ESP32 here
#define MACHINETYPE "ESP32 NodeMCU" // describe your hardware here

// connect ESP32 in STA(tion) mode to your WiFi router ...
#define DEFAULT_STA_SSID          "YOUR_STA_SSID"
#define DEFAULT_STA_PASSWORD      "YOUR_STA_PASSWORD"
// ... but don't let ESP32 to set up its own A(ccess) P(oit)
#define DEFAULT_AP_SSID           ""
#define DEFAULT_AP_PASSWORD       ""

#include "./servers/file_system.h"
#include "./servers/network.h"

#define USER_MANAGEMENT NO_USER_MANAGEMENT  // although NO_USER_MANAGEMENT is selected we still need user_management.h - this is where
#include "./servers/user_management.h"      // web server will get its root directory from, whilest FTP server will accept whatever you type in

#include "./servers/webServer.hpp"
#include "./servers/ftpServer.hpp"

// httpRequestHandler HTTP callback function is no longer needed - web server can read its HTTP response from a HTML file
 
// Callback function that will handle all WS (websocket) requests - they are always called by web browser from HTML (responses or files - see above)

void wsRequestHandler (String& wsRequest, WebSocket *webSocket) { // - must be reentrant!
  if (wsRequest.substring (0, 19) == "GET /streamSamples ") {
    uint16_t buffer [2];
    do {
      delay (100); // 1/10 of a second
      buffer [0] = (int16_t) analogRead (36);
      buffer [1] = (int16_t) analogRead (39);
    } while (webSocket->sendBinary ((byte *) &buffer,  sizeof (buffer))); // push 2, 16 bit unsigned integers (4 bytes) into websocket 
  }
}

void setup () {
  Serial.begin (115200);

  // FFat.format ();                                                    // delete all files if you need to start from scratch
  mountFileSystem (true);                                             // mount (and format if neccessary) file system
 
  startNetworkAndInitializeItAtFirstCall ();                          // initialize network configuration files and start network

  // start web server 
  httpServer *httpSrv = new httpServer (NULL,                         // no HTTP callback function 
                                        wsRequestHandler,             // our WS callback function
                                        8 * 1024,                     // 8 KB stack size is usually enough, if httpRequestHandler or wsRequestHandler use more stack increase this value until server is stable
                                        (char *) "0.0.0.0",           // start HTTP server on all available ip addresses
                                        80,                           // HTTP port
                                        NULL);                        // no firewall callback function
  if (!httpSrv || (httpSrv && !httpSrv->started ())) Serial.println ("[httpServer] did not start.");

  // start FTP server
  ftpServer *ftpSrv = new ftpServer ("0.0.0.0",                       // start FTP server on all available ip addresses
                                     21,                              // controll connection FTP port
                                     NULL);                           // use firewall callback function for FTP server (replace with NULL if not needed)
  if (!ftpSrv || (ftpSrv && !ftpSrv->started ())) Serial.println ("[ftpServer] did not start.");  
}

void loop () {

}

index.html could now look something like this:

<html>
  <body>
    pin 36 <div id='pin36'>...</div> 
    pin 39 <div id='pin39'>...</div>

    <script type='text/javascript'>
      if ('WebSocket' in window) {
        var ws = new WebSocket ('ws://' + self.location.host + '/streamSamples'); // open webSocket connection
        ws.onopen = function () {;};
        ws.onmessage = function (evt) { 
          if (evt.data instanceof Blob) { // Sample readings 
            // receive binary data as blob and then convert it into array¸
            var mySampleArray = null;
            var myArrayBuffer = null;
            var myFileReader = new FileReader ();
            myFileReader.onload = function (event) {
              myArrayBuffer = event.target.result;
              mySampleArray = new Uint16Array (myArrayBuffer); // sample values are now in the array of 16 bit unsigned integers

              // this is where you pass vlues received through websocket into a GUI controls
              for (var i = 0; i < mySampleArray.length; i++) {
                if (i % 2 == 0) document.getElementById('pin36').innerText = mySampleArray [i].toString ();
                else            document.getElementById('pin39').innerText = mySampleArray [i].toString ();
              }
            };
            myFileReader.readAsArrayBuffer (evt.data);
          }
        };
      } else {
        alert ('WebSockets are not supported by your browser.');
      }
    </script>
  </body>
</html>

Source files are attached in ZIP.
[Demo4.zip](https://github.com/BojanJurca/Esp32_oscilloscope/files/6609806/Demo4.zip)

happytm, 2021-06-07:
Thank you for taking so much of your valuable time in order to explain in such a minute detail the concept of servers and the code which make it work. I am sure I will use demo4 for many of my wifi projects in future.

BojanJurca, 2021-06-08:
Enjoy :)

closed 2021-06-08 by BojanJurca