BojanJurca/Esp32_oscilloscope #9: Flash problem

classic-audio, 2021-11-24:
I get the following error:
In file included from c:\users\frede\dropbox\min computer (classic-app-pc)\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\algorithm:60:0,
                 from C:\Users\Frede\Dropbox\Min computer (Classic-App-PC)\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.4\cores\esp32/Arduino.h:142,
                 from sketch\Esp32_oscilloscope.ino.cpp:1:
c:\users\frede\dropbox\min computer (classic-app-pc)\documents\arduinodata\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0\utility:68:28: fatal error: bits/c++config.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.

It is not clear to me where to put the library?

BojanJurca, 2021-11-30:
Just make Esp32_oscilloscope directory and put all the files directly in it. It doesn't seem you have problem with oscilloscope files. I would suggest a clean Arduino installation.

BojanJurca, 2021-11-30:
This article describes how to completely uninstall Arduino (before installing it again).

https://support.arduino.cc/hc/en-us/articles/360021325733-How-to-do-a-complete-uninstall-of-the-Arduino-IDE

Perhaps this could be of some help?

BojanJurca, 2021-12-03:
I'm closing this issue. Please reopen it if it is not resolved.

closed 2021-12-03 by BojanJurca