BojanJurca/Esp32_oscilloscope #13: PArtitionins NodeMCU 32S into FAT

stefesciberras, 2022-06-19:
Could you explain how to get this done? I do not see the option when the board NodeMCU 32S is chosen.

If i chose some of the other options, the partition tool becomes available, but then I cannot compile sketch.

Thanks

BojanJurca, 2022-06-21:
Indeed. It looks like NodeMCU 32s does not support the FAT file system. I have no experience with NodeMCU 32s. Perhaps you can choose some other board that would also work with your NodeMCU 32s?

BojanJurca, 2022-06-22:
You can also place everything in RAM and forget about the file system at all - see the attachment
[Esp32_oscilloscope.ino.zip](https://github.com/BojanJurca/Esp32_oscilloscope/files/8958670/Esp32_oscilloscope.ino.zip)
.

BojanJurca, 2022-06-22:
I added a possibility to choose between FAT and LittleFS file systems. Please download the latest source files. 

When FAT partition scheme is selected than you can define

#define FILE_SYSTEM FILE_SYSTEM_FAT

prior to

#include "file_system.h"

When SPIFFS partition scheme is selected than you can define

#define FILE_SYSTEM FILE_SYSTEM_LITTLEFS

prior to
  
#include "file_system.h"

There is no NodeMCU 32S at my disposal to test it but according to documentation it should be able to run SPIFFS so you should be able to run LittleFS on it.

closed 2022-06-23 by BojanJurca