BojanJurca/Esp32_oscilloscope #15: Watchdog gets triggered when heavily loaded

BojanJurca, 2022-06-25:
One solution may be to increase the CPU frequency of ESP32.

The other solution is to disable the watchdog. 

First

```C++
#include <soc/rtc_wdt.h>
```

then (in setup function)

```C++
void setup () {  
  rtc_wdt_protect_off ();
  rtc_wdt_disable ();
```

If you are using an Esp32 oscilloscope as a stand alone solution this may be perfectly OK. But if it is just a part of your project, disabling the watchdog may not really be what you desire.

actuallydoc, 2022-07-04:
svaka cast da si tole sprogramiru! upam da bom lahk jst kdaj takle kej ustvaru hehe.

BojanJurca, 2022-11-04:
Disable the watchdog if you can afford it by uncommenting the following lies:

  // rtc_wdt_protect_off ();
  // rtc_wdt_disable ();
  // disableCore0WDT ();
  // disableCore1WDT ();
  // disableLoopWDT ();

closed 2022-11-04 by BojanJurca