BojanJurca/Esp32_oscilloscope #16: Displays inverted signal

maarten682, 2022-08-07:
Weird little issue, running the bog standard code (just added wifi credentials) and no matter what ADC1 GPIO i use i always get a high signal from analog read even with nothing connected to the selected pin. When pulling the selected pin down to ground the scope displays 4095, when pulling the pin high to 3.3v i get 0 across the board. 

According to the readme 'analog values from 0 to 4095 <...> correspond to 0 V to 3.3 V' this is the opposite of what it is doing for me. Am i misunderstanding how this is supposed to work?

BojanJurca, 2022-08-07:
Weird, indeed. It seems to me that the issue is not related to the oscilloscope itself, but the problem arises already when reading the analog signal. Maybe you can check if the GPIOs (only 36, 37, 38, 39, 32, 33, 34, 35 can be used for analog reading) are correctly initialized (for example:
   
   analogReadResolution (12); // set 12 bit analog read resolution (default)
   analogSetPinAttenuation (32, ADC_11db); // set attenuation for pin 32 to 2600 mV range (default))

The problem could also arise if you previously performed digitalReads on these GPIOs, which corrupted the GPIO settings.

The idea is to enable correct reading of the GPIOs in oscillospope.html as it makes sense for your project.

maarten682, 2022-08-07:
Ran a quick sketch to test pins;

`void setup() {
  Serial.begin(115200);
}
void loop() {
  Serial.println(analogRead(35));
  delay(5);
}`

Outputs expected 0 and 4095 when hooking pin 35 up to gnd or 3v3 resp.

Reuploaded the scope, same problem. 

I understand i can tweak things but id rather get it working as it is supposed to before i start adding workarounds to make it work as intended.

BojanJurca, 2022-08-08:
Oscilloscope was first intended to work as an embedded system for different kinds of projects, assuming that all GPIOs are already initialized and used as needed. But perhaps you're right, some are using it as a completely stand-alone device so I guess I should initialize 2 digital GPIOs and 2 analog GPIOs in advance and somewhat simplify oscilloscope.html - I'll do this in the afternoon (the similar way as in demo: http://193.77.159.208/oscilloscope.html).

Nevertheless, the results you are getting from analogRead are still weird. Do you perform only analogReading (after startup of ESP32) or do you first digitalRead those GPIOs? If later this may be the cause of false results since digitalRead corrupts GPIO settings.

BojanJurca, 2022-08-08:
Please download the latest oscilloscope.html. I think this would solve your issue.

BojanJurca, 2022-08-11:
I'm closing this issue.

closed 2022-08-11 by BojanJurca