BojanJurca/Esp32_oscilloscope #17: ESP32_cannot_connect_to_WIFI

XingpingGY, 2022-08-26:
I change the wifi ssid and password correctly.
But ESP32 cannot connect to the WiFi.
`[network] starting WiFi
12:00:00.019 -> [       947] [network] [STA] connecting STAtion to router using DHCP
12:00:00.115 -> [      1061] [network] [STA] WiFi client started
12:00:00.115 -> [      1070] [httpServer] started
12:00:00.115 -> [      1071] [ftpServer] started
12:00:02.172 -> [      3112] [network] [STA] disconnected from WiFi
12:00:14.952 -> [     15913] Could not find NTP server by its name.
12:00:14.952 -> [     15914] Could not find NTP server by its name.
12:00:15.000 -> [     15915] Could not find NTP server by its name.`
Hope to know what the problem is. Thank you very much.

XingpingGY, 2022-08-27:
Now I can connect to the WiFi. But the FTP service did not work.
It is said that
`[ftpControlConnection] recv error: 128`
And my windows operating system cannot open the file folder through ftp://+IP this way.

BojanJurca, 2022-08-27:
I'm not sure, it seems that you still experience connection issues. IP address 12:00:00.019 for example is not a valid IP address at all. 

Socket error 128 means:

ENOTSOCK (128)Invalid socket descriptor. An invalid socket descriptor was passed to an API function. 

See: http://www.on-time.com/rtos-32-docs/rtip-32/reference-manual/error-codes.htm

This doesn't really tell me anything right now. I would suggest that you uncomment the following line in setup () function only once:

// FFat.format (); // clear up flash disk to reset everything

then run the scratch again to start from scratch. After this comment it again. Please, send me a console log afterwards.

XingpingGY, 2022-08-30:
Thanks for your patient answer. I did not find the solution to recv:128 and try to open the ftp service via two ways (through cmd command or resource manager of Windows) but failed in both of them. Finally I change a Esp32 and successfully run the program according to the markdown file. Thank you very much!

closed 2022-08-30 by XingpingGY