BojanJurca/Esp32_oscilloscope #18: FTP : 421 file system not mounted

DumoulinPierre, 2022-10-12:
Hi, thanks for this project !

I have an issue at the step where we have to add some files to the ESP32 :

[image]

Thanks,
Pierre.

BojanJurca, 2022-10-12:
Hm, I have just checked it again and it works as it should. Could you, please, give me some additional information like:

- did you change any code on your own?
- did you select a FATFS partition scheme prior to compiling and uploading sketch to Esp32?
- can you send me a console (COM) log?

Thank you.

Bojan

BojanJurca, 2022-10-12:
By the way, if we will not be able to solve the file system problem for some reason there is a workarroud (made mainly for those Esp32 boards that do not have flash disk at all):
- take and compile RAM ONLY version otf the sketch that doesn't require a file system: https://github.com/BojanJurca/Esp32_oscilloscope/blob/master/Esp32_oscilloscope.ino%20RAM%20VERSION

Bojan

DumoulinPierre, 2022-10-12:
Sorry, my bad, I was not doing the Partition Scheme step.. 
But it seems that my board (NodeMCU 32s) doesn't have this option.

I will try your RAM version :)

DumoulinPierre, 2022-10-12:
It's working ! I have the html webpage :) 

thank's a lot for your time and reactivity !

BojanJurca, 2022-10-12:
I'm glad I could help.

B.

closed 2022-10-12 by BojanJurca