BojanJurca/Esp32_oscilloscope #24: Getting Started Issues

bfleg, 2023-11-14:
Hi, looks like a great project, but I'm having a few issues getting going with it - although I'm new to Arduino IDE...

I'm using IDE 2.2.1 and a DFRobot Firebeetle ESP32

When compiling the sketch, it doesn't like the const "OVERFLOW" - some googling suggested it was reserved and should be something else.  I renamed it and the references I could find to OVERFLO - there are some other "warnings" also - is there a recommended IDE version I should be using? Apologies if I missed it.

When the ESP starts up and I try FTP'ing to it, it rejects the request.

Suggestions?

Thanks in advance,

BojanJurca, 2023-11-15:
Hello,

There is no Firebeetle ESP32 at my disposal but I don't think this may be the issue. At first glance it seems that file names on your ESP32 are too long. Maximum number of characters in the full file path is 255.  Can you send me the exact FTP commands you are using? I'll take a look at it today when I come back from work.

There may be a workaround meanwhile. You can also use the oscilloscope without a file system. Just comment out the line

 // #define FILE_SYSTEM    FILE_SYSTEM_FAT

in the file Esp32_oscilloscope.ino.

bfleg, 2023-11-15:
Hi, thanks... Yes, I commented out the FILE_SYSTEM part (a lot more errors in compile with it in!)

It turns out the ESP comes up and I was able to get to the webserver /oscilloscope page without FTP'ing, but seems it also take a long time to come up too.  Is the file length just on the FTP, or on the project as a whole in the IDE?  Essentially, as it is (/was) the FTP service doesn't appear to be up at all - I can't open an FTP connection to it at all (tried with winSCP (FTP mode) and also command line FTP.

BojanJurca, 2023-11-15:
If you are using the oscilloscope without the file system, FTP service is not mounted. This wouldn't make sense. 

The file length limitation is just on ESP32's site. You should FTP the necessary files to /var/www/html directory (ESP32's site).

The server should be ready as soon as ESP32 gets its IP from the router.

```
C:\esp32_oscilloscope>ftp YOUR-ESP32-IP
Connected to 10.0.0.3.
220-ESP32 FTP server - everyone is allowed to login
User (10.0.0.3:(none)):
331 enter password
Password:
230 logged on, use "/" to refer to your home directory "/"
ftp> put android-192-osc.png /var/www/html/android-192.png
226 /var/www/html/android-192-osc.png transfer complete
ftp> put apple-180-osc.png /var/www/html/apple-180.png
226 /var/www/html/apple-180-osc.png transfer complete
ftp> put oscilloscope.html /var/www/html/oscilloscope.html
226 /var/www/html/oscilloscope.html transfer complete
ftp>
```

bfleg, 2023-11-15:
...it won't connect to the FTP service at all.  Given the instructions, I hadn't tried browsing to it until much later, but when I did, the oscilloscope page came up anyhow.
Ok, so if it's not expected to be able to FTP when there's no FS, this would make a lot more sense I guess!  Now I just need to see why it takes so long to come online or not come on at all...

BojanJurca, 2023-11-15:
If you areconnection your ESP32 in STAtion mode (to the router) please check/modify the following settings:

```
// 3. NETWORK:     #define how ESP32 will use the network
    // STA(tion)
    // #define how ESP32 will connecto to WiFi router
    // this information will be written into /etc/wpa_supplicant/wpa_supplicant.conf file if file_system.h will be included
    // if these #definitions are missing STAtion will not be set up
    #define DEFAULT_STA_SSID                          "YOUR_STA_SSID"       // <- replace with your information
    #define DEFAULT_STA_PASSWORD                      "YOUR_STA_PASSWORD"   // <- replace with your information
    // the use of DHCP or static IP address wil be set in /network/interfaces if file_system.h will be included, the following is information needed for static IP configuration
    // if these #definitions are missing DHCP will be assumed
    // #define DEFAULT_STA_IP                            "10.18.1.200"      // <- replace with your information
    // #define DEFAULT_STA_SUBNET_MASK                   "255.255.255.0"    // <- replace with your information
    // #define DEFAULT_STA_GATEWAY                       "10.18.1.1"        // <- replace with your information
    // #define DEFAULT_STA_DNS_1                         "193.189.160.13"   // <- replace with your information
    // #define DEFAULT_STA_DNS_2                         "193.189.160.23"   // <- replace with your information
```

bfleg, 2023-11-15:
...yup - that was done.  It's odd, it doesn't seem to come up and appear in the router, but it seems to come online a lot later (1hour+) might try static IP

BojanJurca, 2023-11-15:
I tried to reproduce your problems with the following settings (Tools | ...):

- board: DFRRobot Firebeetle 2 ESP32
- scheme: 4 MB with ffat (1.2 MB APP / 1.45 MB FATFS)

1. your board does not have i2s interface, so comment out: // #define USE_I2S_INTERFACE in oscilloscope.h Analog sampling will still work although it will not be as fast and as precise as with i2s interface.
This is why the function oscReader_analog_1_signal_continuous (in oscilloscope.h) doesn't compile. You may freely delete it.

You may also comment out // #include <soc/rtc_wdt.h>  in Esp32_oscilloscope.ino and the following as well - it is possible, though, that your Firebeetle will reset if analog sampling in high frequencies, though.

```
  // disable watchdog if you can afford it - watchdog gets occasionally triggered when loaded heavily
  /*
  rtc_wdt_protect_off ();
  rtc_wdt_disable ();
  disableCore0WDT ();
  disableCore1WDT ();
  disableLoopWDT ();
  */
```

The watchdog is obviously implemented differently or not at all for S3 boards. I'm sorry for the inconvenience, I could not anticipate all the possible problems with different ESP32 boards.

Then you should get the code compiled without errors, both, with or without a file system. If no file system is used then the .html file is served from memory. This is probably not bad at all but if an oscilloscope is used as embedded system, this would use some of the memory that may be needed elsewhere.

```
Sketch uses 788129 bytes (60%) of program storage space. Maximum is 1310720 bytes.
Global variables use 57184 bytes (17%) of dynamic memory, leaving 270496 bytes for local variables. Maximum is 327680 bytes.
```
or 

```
Sketch uses 780553 bytes (59%) of program storage space. Maximum is 1310720 bytes.
Global variables use 56940 bytes (17%) of dynamic memory, leaving 270740 bytes for local variables. Maximum is 327680 bytes.
```

I'm afraid this is as much as I can do, since I do not have any S3 board at my disposal. I should buy one, I suppose ...

bfleg, 2023-11-16:
Thanks very much - I've removed all the stuff you mentioned but sadly, it's still hit and miss as to whether the networking parts come up - mostly they don't.  The need isn't urgent (wanted to look at PWM fan tach / pwm signal to debug a different problem), so I may abandon for now.  Thanks for your help anyhow.

BojanJurca, 2023-11-16:
No problem. I ordered ESP32 S2, S3 and C3 boards anyway and will fix this issue once they arrive from China.

BojanJurca, 2023-12-10:
Hi. I got Oscilloscope working on ESP32 S3 and will publish the updated code soon. But I'm having problems getting demo PWM working myself. It seems that  ESP32 S3 may have issues with it. I didn't have such problems with any other ESP32 board so far

bfleg, 2023-12-10:
Hi, thanks very much for the update - I'm not too sure what an "S3" is in reality, I've just been using the DFRobot Firebeetle DFR0478 - it would be awesome if it's possible to use it. Thanks again.

BojanJurca, 2023-12-23:
I uploaded support for S2, S3 and C3 boards. I am somewhat confused about which board to use with DFRobot Firebeetle DFR0478. I ordered one myself just to be able to try it but it hasn't arrived yet.

Nevertheless, I tried compiling Oscilloscope for Firebeetle 2 ESP32-E and Firebeetle-ESP32 boards. In both cases it now compiles right away, without needing to modify the code anyhow.

As I said, I'm still not able to actually try it on the real DFRobot Firebeetle DFR0478 board.

bfleg, 2023-12-23:
Thanks very much, I’ll give it a go and let you know if it’s working!

BojanJurca, 2024-01-11:
I've got my DFRobot Firebeetle DFR0478 and compiled code for the Firebeetle 2 ESP32-E board. Everything works as expected. I'm closing this issue.

closed 2024-01-11 by BojanJurca