BojanJurca/Esp32_oscilloscope #28: https Server

nebula7219, 2024-04-06:
Is it possible to integrate an https server so that I can use it as an iframe in an IoT project?

thanks
ralph

BojanJurca, 2024-04-06:
I'm sorry I don't understand your question.

If your question is if you can use https server instead of http, well, you
can, but you will have to do the porting to another Web server yourself.
Which https server did you have in mind? One that I like is:
https://github.com/fhessel/esp32_https_server, it supports Websockets and
multitasking. The performance will probably drop for ESP32 will also have
to do the encryption.

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

BojanJurca, 2024-04-06:
But if you only want to use some additional stuff beside Oscilloscope.html,
then the answer is of course. Just FTP your files to /var/www/html
directory. The complete project (HTTP, FTP and Telnet servers +
Oscilloscope) is here:
https://github.com/BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2024-04-17 by BojanJurca