BojanJurca/Esp32_oscilloscope #30: Compilation error: expected nested-name-specifier

chriskuku, 2024-04-18:
I'm trying to build `Esp32_oscilloscope.ino` using Arduino-IDE 2.3.2 and I'm getting somthing like this:
(output has scrolled out of the output window, so I cannot see where the error initially occurs.

``

      |             ^~~~~~~~
/Users/kuku/Esp32_oscilloscope/fsString.h:46:13: note:   candidate expects 1 argument, 2 provided
/Users/kuku/Esp32_oscilloscope/fsString.h:44:13: note: candidate: 'fsString<N>::fsString() [with unsigned int N = 300]'
   44 |             fsString () {}                                      // for declarations like fsString<15> a;
      |             ^~~~~~~~
/Users/kuku/Esp32_oscilloscope/fsString.h:44:13: note:   candidate expects 0 arguments, 2 provided

exit status 1

Compilation error: expected nested-name-specifier

``

BojanJurca, 2024-04-18:
I have just tried compiling it with Arduino IDE 2.3.2 - with no problems. Which board did you choose?

chriskuku, 2024-04-19:
> I have just tried compiling it with Arduino IDE 2.3.2 - with no problems. Which board did you choose?

ESP32 Dev Module

Maybe my installed libraries are in conflict with those in ../libraries?

chriskuku, 2024-04-19:
NB: I have undef'd filesystem (FWIW):

```
    // COMMENT THIS DEFINITION OUT IF YOUR ESP32 DOES NOT HAVE A FLASH DISK
  //  #define FILE_SYSTEM    FILE_SYSTEM_LITTLEFS // FILE_SYSTEM_FAT  // the file system must correspond to Tools | Partition scheme setting: FILE_SYSTEM_FAT (for FAT partition scheme), FILE_SYSTEM_LITTLEFS (for SPIFFS partition scheme) or FILE_SYSTEM_SD_CARD (if SC card is attached)
                                                // FAT file system can be bitwise combined with FILE_SYSTEM_SD_CARD, like #define FILE_SYSTEM (FILE_SYSTEM_FAT | FILE_SYSTEM_SD_CARD)
```

chriskuku, 2024-04-19:
> ESP32 Dev Module has a built-in flash disk, so the sketch can be compiled with file system enabled, perhaps you can try this option?
> 
> Although it shouldn't be a problem compiling it without the file system as well. I'll take a look at it in the evening.

OK, but ESP32 Dev was only chosen by me since I didn't know what else I should choose. Actually I have a CYD2USB which has a socket for an SD card. Will get me a card anyway.

I'm not sure which WiFi.h is taken but I find one in
```
$ ls
AP.cpp		WiFi.cpp	WiFiAP.cpp	WiFiClient.h	WiFiGeneric.h	WiFiMulti.h	WiFiSTA.h	WiFiScan.h	WiFiType.h
STA.cpp		WiFi.h		WiFiAP.h	WiFiGeneric.cpp	WiFiMulti.cpp	WiFiSTA.cpp	WiFiScan.cpp	WiFiServer.h	WiFiUdp.h
$ vi WiFi.cpp
$ pwd
/Users/kuku/Library/Arduino15/packages/esp32/hardware/esp32/3.0.0-rc1/libraries/WiFi/src
$ 

```

chriskuku, 2024-04-19:
Update: I removed ESP32 3.0.0-rc and installed ESP32 2.0.11 and compilation ran through.
Despite from this message : 

```
In file included from /Users/kuku/Esp32_oscilloscope/Esp32_oscilloscope.ino:130:
/Users/kuku/Esp32_oscilloscope/oscilloscope.h:58:25: note: #pragma message: Oscilloscope will use I2S interface (for monitoring a single analog signal) and adc1_get_raw (for monitoring double analog signals).
         #pragma message "Oscilloscope will use I2S interface (for monitoring a single analog signal) and adc1_get_raw (for monitoring double analog signals)."
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Sketch uses 847605 bytes (64%) of program storage space. Maximum is 1310720 bytes.
Global variables use 55424 bytes (16%) of dynamic memory, leaving 272256 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.usbserial-14120
Connecting.....
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:b6:1f:b8:5f:b0
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x000e0fff...
Compressed 18992 bytes to 13112...
Writing at 0x00001000... (100 %)
Wrote 18992 bytes (13112 compressed) at 0x00001000 in 0.6 seconds (effective 257.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.1 seconds (effective 448.3 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 656.9 kbit/s)...
Hash of data verified.
Compressed 853360 bytes to 529675...
Writing at 0x00010000... (3 %)
Writing at 0x0001c9fe... (6 %)
Writing at 0x00029ef2... (9 %)
Writing at 0x000340b2... (12 %)
Writing at 0x0004191d... (15 %)
Writing at 0x00046c0a... (18 %)
Writing at 0x0004c0c8... (21 %)
Writing at 0x00051814... (24 %)
Writing at 0x000571ec... (27 %)
Writing at 0x0005c33f... (30 %)
Writing at 0x000615d7... (33 %)
Writing at 0x000668a5... (36 %)
Writing at 0x0006bc92... (39 %)
Writing at 0x00070f02... (42 %)
Writing at 0x00076309... (45 %)
Writing at 0x0007b536... (48 %)
Writing at 0x000810b4... (51 %)
Writing at 0x00086825... (54 %)
Writing at 0x0008bab4... (57 %)
Writing at 0x00091049... (60 %)
Writing at 0x0009629b... (63 %)
Writing at 0x0009b9ad... (66 %)
Writing at 0x000a10fa... (69 %)
Writing at 0x000a6b0a... (72 %)
Writing at 0x000ac840... (75 %)
Writing at 0x000b1e29... (78 %)
Writing at 0x000b9eef... (81 %)
Writing at 0x000c12b2... (84 %)
Writing at 0x000c89a8... (87 %)
Writing at 0x000cdd1d... (90 %)
Writing at 0x000d374c... (93 %)
Writing at 0x000d8dfe... (96 %)
Writing at 0x000de6f6... (100 %)
Wrote 853360 bytes (529675 compressed) at 0x00010000 in 13.5 seconds (effective 503.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```

I was able to upload it to the device (CYD). The screen remains black which may be normal  if it runs over the net.

So the above error (reason of my post) is due to some quirks in ESP32 3.0.0-rc

BojanJurca, 2024-04-19:
I'm sorry, I can not replicate your error. The code compiles right away with or without a file system. 

Perhaps you can try installing the Arduino IDE from scratch and download ESP_oscillocsope again?

chriskuku, 2024-04-19:
> I'm sorry, I can not replicate your error. The code compiles right away with or without a file system.
> 
> Perhaps you can try installing the Arduino IDE from scratch and download ESP_oscillocsope again?

Thanks for doing the test. Did you try ESP32-3.0.0-rc?
That’s the configuration that triggers the error.

BojanJurca, 2024-04-19:
No, unfortunately I do not have this board installed.

Tozzi89, 2024-05-28:
I get the same issue. Also have ESP32-3.0.0 (Final, not RC). Using ESP32 C3 Super Mini with chipset fn4p3m4490.
Installing ESP32-2.0.17 I get new error:

oscilloscope.h:860:23: note: suggested alternative: 'i2s_set_clk'
                 err = i2s_set_adc_mode (ADC_UNIT_1, adcchannel1);
                       ^~~~~~~~~~~~~~~~
                       i2s_set_clk

exit status 1

Compilation error: 'I2S_MODE_ADC_BUILT_IN' was not declared in this scope

BojanJurca, 2024-05-28:
Thank you for letting me know. 

Unfortunately I'm away now and will try to reproduce the error as soon as I come back.

ria8651, 2024-06-12:
Downgrading to 2.0.17 worked for me.

> Compilation error: 'I2S_MODE_ADC_BUILT_IN' was not declared in this scope

I had this issue too and just had to comment out `#define USE_I2S_INTERFACE` as my board doesn't have one. It'd probably be a good idea to use `MACHINETYPE` to disable `USE_I2S_INTERFACE` based on [this](https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32s2/api-reference/peripherals/i2s.html#overview-of-all-modes).

BojanJurca, 2024-06-12:
Thank you for your suggestion. I'll take a look at it to see what I can do.

BojanJurca, 2024-06-16:
I have just uploaded updated files that work with IDF 5.x. I think this will resolve your problem.

Tozzi89, 2024-06-16:
Awesome, I will try it out in a few days!

ria8651, 2024-06-18:
Works perfectly

closed 2025-04-08 by BojanJurca