BojanJurca/Esp32_oscilloscope #31: Compilation error: "was not declared in this scope;"

axlFThn, 2024-05-31:
When compiling this project for an Adafruit ESP32 feather board, I get hundreds of was "not declared in this scope;" errors.

BojanJurca, 2024-06-01:
Thank you for letting me know.

Unfortunately I'm away now and will try to reproduce the error as soon as I come back.

BojanJurca, 2024-06-08:
Compile-time errors are caused by migration from IDF 4.4 to 5.0. I'm currently working on this but can't really predict when it will be completed.

BojanJurca, 2024-06-16:
I have just uploaded updated files that work with IDF 5.x. I think this will resolve your problem.

paramoshkinandrew, 2024-06-17:
I got similar error with `esp_netif_next_unsafe` in `netwk.h`. Renaming `esp_netif_next_unsafe` to `esp_netif_next` did the trick and sketch compiled. @axlFThn that could be the one you get too :)

BojanJurca, 2025-04-08:
I'm sorry I wasn't checking the issues for a while. I think it may not be relevant any more. EPS32's functions have changed, but not for all board versions. Most of them upgraded from  esp_netif_next to  esp_netif_next_unsafe, but not all of them. So if the code does not compile with  esp_netif_next_unsafe just change the call back to  esp_netif_next. It should work.

closed 2025-04-08 by BojanJurca