BojanJurca/Esp32_oscilloscope #32: Average sampling rate?

Algtronic, 2024-08-24:
You don't mention sampling rate in your documentation. Can you please supply me with this information?

BojanJurca, 2024-08-24:
Well, it depends. Mostly on horizontal frequency. The easiest way to figure it out is by checking oscilloscope.html file on how startCommand is constructed (from lines 464 on).

Basically this means that it can be changed according to your wishes as long as you don't exceed screen buffer which is:

 max 745 (continuous analog signal) samples per screen,
 max 372 (1 signal) samples per screen,
 max 247 (2 signals) samples per screen.

Algtronic, 2024-08-24:
I mean per second. But thank you for additional information!

BojanJurca, 2024-08-24:
"every 2500 ms" for example actually means 400 / s. 

But please note that this sampling rate is only guaranteed when I2S (continuous DMA) sampling is in place. Otherwise samples may be delayed if the processor has other things to do. You can see on the screen exactly when a particular sample has been taken.

closed 2024-09-14 by Algtronic