BojanJurca/Esp32_oscilloscope #33: Unable to transfer file

piul200, 2024-08-24:
[image]
He didn't seem to respond

BojanJurca, 2024-08-24:
I can not see in your console that the file system is mounted. The output should look like:

```
[fileSystem] LittleFS mounted

[network] reading /network/interfaces ... OK, using DHCP
[network] reading /etc/wpa_supplicant/wpa_supplicant.conf ... OK
[network] reading /etc/dhcpcd.conf ... OK
[network] reading /etc/hostapd/hostapd.conf ... OK, 
[network][STA] connecting to: ***
[network][STA] is using DHCP
[network][STA] interface ready
[network][STA] client started
[network][STA] disconnected
[network][STA] connected to ****[network][STA] got IP address ***
```

Does your board have flash disks? Did you by some chance excluded FILE_SYSTEM in Esp32_servers_config.h?

piul200, 2024-08-24:
My board is esp32-s3-n16r8 with a 16M flash disk. For your source code, I just changed my wifi name and password, and selected `FILE_SYSTEM_LITTLEFS` as the file system
[image]

BojanJurca, 2024-08-24:
This is strange. You should get the information whether the file system has been mounted or did it fail.

I'll check if I still have a ESP32 S3 board at my disposal and try to compile the code for it.

Is there a way you check if the file system is mounted?
Did you choose the SPIFFS partition schema?

Well, you can also decide not to #define FILE_SYSTEM at all. In tiss case oscilloscope will serve a .html page from RAM.

piul200, 2024-08-24:
I don't know much about SPIFFS, including the way to check it. As far as I know, the esp32's default partition table `partitions.csv` does not contain partitions for SPIFFS.

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size,   Flags
nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 1M,
```

Also, if `#define FILE_SYSTEM` is disabled,  `FILE_SYSTEM` seems to default to `FILE_SYSTEM_FAT`.

file_system.h:
```
#ifndef FILE_SYSTEM
      #define FILE_SYSTEM FILE_SYSTEM_FAT // by default
#endif
```

I tried two things:
1. Use `FILE_SYSTEM_FAT`     This causes the cmd output `421 file system not mounted`
2. Disable `#define FILE_SYSTEM` at all     This leads to many compilation errors

BojanJurca, 2024-08-24:
I have just successfully compiled and tested oscilloscope on ESP32 S3 with the following Tools settings:

[image]

Console output is as expected:

```
[fileSystem] LittleFS mounted

[network] reading /network/interfaces ... OK, using DHCP
[network] reading /etc/wpa_supplicant/wpa_supplicant.conf ... OK
[network] reading /etc/dhcpcd.conf ... OK
[network] reading /etc/hostapd/hostapd.conf ... OK, 
[network][STA] connecting to: ***
[network][STA] is using DHCP
[network][STA] interface ready
[network][STA] client started
[network][STA] disconnected
[network][STA] connected to ***
[network][STA] got IP address ***
```

piul200, 2024-08-25:
Following your Settings, I get a new log.
I tried again to transfer the file using ftp, but was unsuccessful.
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4208eade
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[       226] [file system] mounted.
[       227] [cronDaemon] starting ...
[       250] [cronDaemon] reading /etc/ntp.conf
[       291] [cronDaemon] reading /etc/crontab
[       320] [cronDaemon][cronAdd] invalid cronTabLine: 
[       320] [cronDaemon] started
[       396] [network] starting WiFi
[       396] [network] [STA] connecting STAtion to router using DHCP
[       441] [network] [STA] WiFi client started
[       458] [httpServer] started
[       459] [ftpServer] started
[       473] [network] [STA] connected to WiFi ***
[       487] [network] [STA] got IP address: 192.168.2.196
```

BojanJurca, 2024-08-25:
Honestly, I'm running out of ideas.

- You can still try excluding file system (comment #define FILE_SYSTEM line), this way fileSystem.hpp will not get included at all.

- the other possibility is to use Telnet instead of FTP. Include telnetServer.hpp and then run it:

```
    #ifdef __TELNET_SERVER__
        telnetServer *telnetSrv = new (std::nothrow) telnetServer (telnetCommandHandlerCallback, "0.0.0.0", 23, NULL);
        if (telnetSrv && telnetSrv->state () == telnetServer::RUNNING) {
            cout << "[telnetServer] started" << endl;
        } else {
            dmesgQueue << "[telnetServer] did not start";
            cout << "[telnetServer] did not start" << endl;            
        }
    #endif
```

If you telnet to your ESP32 later you can then
```
# cat > /var/www/html/oscilloscope.html
```
... and paste the content of a file there.

piul200, 2024-08-25:
Okay, I think there might be something wrong with my board that I should check.
Thank you for your suggestion, I think it is a great project, goodbye

closed 2024-08-25 by piul200