BojanJurca/Esp32_oscilloscope #34: Compilation error: 'esp_netif_next_unsafe' was not declared in this scope

PaulSchulz, 2024-10-14:
Greetings, I got the compilation error when trying to compile the Esp32_oscilloscope code for a Heltec Wifi Kit 32 V3.
(This is an ESP32S3 board, without any PSRAM)

The code successfully builds and installs on the following:
- Lilygo T-Display (T Display PiICOC3) (ESP32C3); and an 
- M5ATOM Echo (ESP-PICO-D4)

Using Arduino IDE v2 (arduino-ide_2.3.3_Linux_64bit.AppImage) on Ubuntu 24.04

```
In file included from /home/paul/Documents/git/Esp32_oscilloscope/servers/time_functions.h:31,
                 from /home/paul/Documents/git/Esp32_oscilloscope/Esp32_servers_config.h:91,
                 from /home/paul/Documents/git/Esp32_oscilloscope/Esp32_oscilloscope.ino:27:
/home/paul/Documents/git/Esp32_oscilloscope/servers/netwk.h: In function 'void startWiFi()':
/home/paul/Documents/git/Esp32_oscilloscope/servers/netwk.h:791:30: error: 'esp_netif_next_unsafe' was not declared in this scope
         esp_netif_t *netif = esp_netif_next_unsafe (NULL);
                              ^~~~~~~~~~~~~~~~~~~~~
/home/paul/Documents/git/Esp32_oscilloscope/servers/netwk.h:791:30: note: suggested alternative: 'esp_netif_next'
         esp_netif_t *netif = esp_netif_next_unsafe (NULL);
                              ^~~~~~~~~~~~~~~~~~~~~
                              esp_netif_next
In file included from /home/paul/Documents/git/Esp32_oscilloscope/Esp32_servers_config.h:106,
                 from /home/paul/Documents/git/Esp32_oscilloscope/Esp32_oscilloscope.ino:27:
/home/paul/Documents/git/Esp32_oscilloscope/servers/oscilloscope.h: At global scope:
/home/paul/Documents/git/Esp32_oscilloscope/servers/oscilloscope.h:66:25: note: #pragma message: Oscilloscope will use I2S interface (for monitoring a single analog signal) and adc1_get_raw (for monitoring double analog signals).
         #pragma message "Oscilloscope will use I2S interface (for monitoring a single analog signal) and adc1_get_raw (for monitoring double analog signals)."
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/paul/Documents/git/Esp32_oscilloscope/Esp32_servers_config.h:106,
                 from /home/paul/Documents/git/Esp32_oscilloscope/Esp32_oscilloscope.ino:27:
/home/paul/Documents/git/Esp32_oscilloscope/servers/oscilloscope.h: In function 'void oscReader_analog_1_signal_i2s(void*)':
/home/paul/Documents/git/Esp32_oscilloscope/servers/oscilloscope.h:891:75: error: 'I2S_MODE_ADC_BUILT_IN' was not declared in this scope
                     .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_RX | I2S_MODE_ADC_BUILT_IN),
                                                                           ^~~~~~~~~~~~~~~~~~~~~
/home/paul/Documents/git/Esp32_oscilloscope/servers/oscilloscope.h:920:23: error: 'i2s_set_adc_mode' was not declared in this scope
                 err = i2s_set_adc_mode (ADC_UNIT_1, adcchannel1);
                       ^~~~~~~~~~~~~~~~
/home/paul/Documents/git/Esp32_oscilloscope/servers/oscilloscope.h:920:23: note: suggested alternative: 'i2s_set_clk'
                 err = i2s_set_adc_mode (ADC_UNIT_1, adcchannel1);
                       ^~~~~~~~~~~~~~~~
                       i2s_set_clk

exit status 1

Compilation error: 'esp_netif_next_unsafe' was not declared in this scope
```

BojanJurca, 2024-10-15:
Hello,

'esp_netif_next_unsafe' problem (with some boards) has been solved here: https://github.com/BojanJurca/Esp32_oscilloscope/issues/31. So please, just replace 'esp_netif_next_unsafe' with 'esp_netif_next' function.

'i2s_set_adc_mode' problem occurs because your board does not have an appropriate I2S interface. Oscilloscope will still run on this board if you comment out the '#define USE_I2S_INTERFACE' line in Esp32_servers_config.h. Single channel analog sampling will not be that fast and accurate though.

PaulSchulz, 2024-10-19:
Is the fix foir #31 going into the main code? (Thank you for letting me know.)

BojanJurca, 2024-10-19:
esp_netif_next will be used in the next release.

According to ESP-IDF Programming Guide (https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/network/esp_netif.html) esp_netif_next is deprecated and Espressif suggests using a fasest function esp_netif_next_unsafe. But there are some boards where esp_netif_next_unsafe is not supported. I haven't come across a board where esp_netif_next would not be supported though.

BojanJurca, 2025-04-08:
I'm sorry I wasn't checking the issues for a while. I think it may not be relevant any more. EPS32's functions have changed, but not for all board versions. Most of them upgraded from  esp_netif_next to  esp_netif_next_unsafe, but not all of them. So if the code does not compile with  esp_netif_next_unsafe just change the call back to  esp_netif_next. It should work.

closed 2025-04-08 by BojanJurca