BojanJurca/Esp32_oscilloscope #35: Network alway disconneted

HilalBisrie, 2024-10-17:
I’m facing an issue where my ESP32 connects to WiFi but disconnects immediately after. Could you provide a solution to resolve this problem? I've also added serial begin in void setup and change network. Please help me sir
[image]

BojanJurca, 2024-10-17:
It looks like you're having problems with WiFi connection. I suggest

1. checking DEFAULT_STA_SSID and DEFAULT_STA_PASSWORD in Esp32_servers_config.h

2. temporarily uncomment // fileSystem.formatLittleFs ();  line in Esp32_oscilloscope.ino and run the code once to format flash disk and start from the scratch.

3. comment out fileSystem.formatLittleFs (); to prevent formatting at each restart.

HilalBisrie, 2024-10-17:
I've done it, but still disconnected sir. Idk what's the main problem in my case, because i've try using the same config network for other project is working. So,  should I change the tz in server.ino sirr?
[image]

BojanJurca, 2024-10-17:
TZ (timezone) is irrelevant for the oscilloscope itself and has nothing to do with WiFi. 

I can see that you are connecting to Samsung SSID. Is this your phone? Is it possible you didn't turn your phone's hotspot on?

BojanJurca, 2024-10-17:
It seems that you are compiling a test/development Arduino sketch "servers.ino".

Please, use "Esp32_oscilloscope.ino" and define your settings in "Esp32_servers_config.h" before compiling.

BojanJurca, 2024-10-17:
But if you are interested in the server project, please visit https://github.com/BojanJurca/Multitasking-Esp32-HTTP-FTP-Telnet-servers-for-Arduino.

In this case you will most likely want to change TZ to your timezone which you can find here: https://github.com/nayarsystems/posix_tz_db/blob/master/zones.csv

HilalBisrie, 2024-10-18:
My issue was resolved by changing the router SSID from 'Samsung galaxy s22' to 'Ywach'. It turns out that the network configuration only reads the first word of the SSID, which caused the problem initially.

Thank you so much, you're very kind sir

[image]

closed 2025-04-08 by BojanJurca